thu-ml/RoboticsDiffusionTransformer #14: Fine-Tuning in Relative Action Space

lakomchik, 2024-11-06:
I woule like for fine-tune RDT in a relative action space and have a question regarding the best method for mapping actions and proprieception.

Question: For fine-tuning a model in relative action space, would it be preferable to:
- Map relative joint positions directly into a unified action space, as per existing guidelines?
- Normalize values (e.g., scaling from -1 to 1) and then project these normalized values into the unified action space?

Using relative actions results in a smaller range for proprioception and action values. I’m curious if normalizing these values could help them better align with the model’s expected action space.

csuastt, 2024-11-08:
Map them into the velocity slots in the unified action space (e.g., delta eef positions should be in eef position velocities). You could do normalization when fine-tuning.

lakomchik, 2024-11-11:
@csuastt Thank you for your answer!

budzianowski, 2024-11-14:
@csuastt - all preprocessing scripts are using `eef_delta_pos_x` (see https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/data/preprocess_scripts/bridgev2.py#L128) and all the other preprocess_scripts for OXE.
I can't find the place where these slots are formatted to `eef_vel_x`?

alik-git, 2024-11-14:
@csuastt  Follow up question: In the example above you can see the model is predicting eef_ang_x, y, z, w for a quaternion, but I don't see a way to map these **directly** to velocities, because in the `STATE_VEC_IDX_MAPPING` all the angular velocities are roll pitch raw only, see here

https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/configs/state_vec.py#L62

Could you please clarify which indices and how exactly you are mapping the quaternion eef_ang_x, y, z, w to `STATE_VEC_IDX_MAPPING`. That would be greatly appreciated. Thank you!

csuastt, 2024-11-15:
@alik-git @budzianowski Sorry, it is our mistake:( In the current implementation, we do not use any action in TFDataset. We use future states instead. To use action, you may need to do some modifications:

1. In this line, remove the function converting RPY to quat, it was a mistake and we forgot to delete:

https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/data/preprocess_scripts/bridgev2.py#L115

The original action is already delta RPY, which is the angle velocity.

2. you may need to modify the follow-up preprocessing script to make the producer generate the action instead of future states. See this readme:

https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/docs/pretrain.md?plain=1#L242

budzianowski, 2024-11-15:
Thanks for the prompt reply, this is very helpful! One more question - the model used in the demos from the paper is also following the this logic or the finetuning was performed with the modified logic?

ethan-iai, 2024-11-16:
To clarify, in the demos mentioned in the paper, we predict the future states following the same logic. Please let me know if you’d like further details!

budzianowski, 2024-11-16:
@ethan-iai Thanks for helpful explanation! If that's the case I'm still puzzled by the finetuning agilex setup where the actions are used?

alik-git, 2024-11-16:
@ethan-iai @csuastt I just want to clarify when you say "predict the actions" are you saying that the neural network model directly outputs action deltas as the logits? or are you saying that the model directly outputs future states, and then you use that to manually compute the action deltas (future_state - current_state = action_deltas)?

The reason I ask is that during pretraining the model predicts future states as the logits (please correct me if that's wrong), so why not keep that consistent during fine tuning as well? 

Just for context: we are trying to evaluate RDT on controlling a widowx robot arm. We are wondering if during our finetuning, it would be better to have the gt labels be the future states, and then compute the action deltas manually during deployment, or finetune with action deltas directly as the gt_labels. My naive assumption was that finetuning with action deltas directly would be worse since the model has to relearn more stuff (due to differences in scale and representation, e.g., smaller ranges for action deltas compared to joint positions).

But if you finetuned directly on action deltas (and empirically found it to be better) then we should reconsider our approach of finetuning on future states. Sorry for the long question, I just wanted to be extra clear in what the confusion is. Thank you for your time in answering all these questions, we greatly appreciate it!

ethan-iai, 2024-12-17:
@alik-git Sorry for the late reply! Below are my responses to your questions:

**Q1: Action Prediction Clarification**

> Is the model predicting action deltas directly as logits, or is it predicting future states and then manually computing action deltas (future state - current state = action deltas)?

**A1:** To clarify, we predict the future states of the leader arms of ALOHA (which termed as action), while the states of the follower arms serve as the input robot proprioception (state).

**Q2: Pretraining vs Fine-tuning Consistency**

> Since the model predicts future states during pretraining, why not keep this approach consistent during fine-tuning?

**A2:** From the perspective of A1, I believe it’s more consistent with the pretraining approach to continue predicting future states during fine-tuning.

**Q3: GT Labels during Fine-tuning**

> Would it be better to use future states as the ground truth labels during fine-tuning and manually compute action deltas during deployment, or should action deltas be used as the ground truth labels directly for fine-tuning?

**A3:** In my view, it would be better to use future states as the ground truth during fine-tuning. During deployment, it would be ideal to directly set the arm to the predicted states. (I’m not familiar with windowx arm, so I’m unsure whether this is feasible with that specific hardware.)

**Q4: Impact of Using Action Deltas for Fine-tuning**

> If action deltas are used as the ground truth during fine-tuning, would it be worse because the model might need to relearn due to differences in scale and representation (e.g., smaller ranges for action deltas compared to joint positions)?

**A4:** It would likely be worse to use absolute value slots for action deltas. It’s generally better to represent deltas using velocity-based slots instead of position-based ones.

**Q5: Empirical Evidence on Fine-tuning with Action Deltas**

> If fine-tuning directly on action deltas has been empirically found to perform better, should we reconsider the approach of fine-tuning on future states?

**A5:** This is still an open question and remains to be explored further. To the best of our knowledge, models (e.g., Octo & OpenVLA) tend to predict deltas for end-effector positions (EEF), while other models (e.g., ACT) predict future states for joint positions. However, one thing is certain: when using low-end (non-industrial level) robot arms, it’s generally better to predict future states.

closed 2025-01-13 by lakomchik