thu-ml/RoboticsDiffusionTransformer #15: Fine-tuning the model using data in RLDS format

prilgrim-whtie, 2024-11-07:
Thank you for your outstanding contribution in the field of bimanual manipulation, I am a beginner in this field and I would like to ask how do I go about fine-tuning the model if the dataset I am using is in RLDS format. The fine tuning guide for this project seems to use the hdf5 storage format. If I use the dataset in Open X-Embodiment (RLDS format) for fine-tuning, and I use the pre-training pipeline, how do I modify it, and is it possible to write a fine-tuning guideline about this, because it seems that pretrain.sh contains pre-training commands, and it does not contain the fine-tuning commands!

csuastt, 2024-11-08:
You could modify [pretrain_datasets.json](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/configs/pretrain_datasets.json) and [pretrain_sample_weights.json](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/configs/pretrain_sample_weights.json) to remove all the irrelevant datasets, and directly use `finetune.sh` by removing `--load_from_hdf5` and replacing `--dataset_type="finetune"` with `--dataset_type="pretrain"`. Don't forget to run producer first!

prilgrim-whtie, 2024-11-08:
Thanks for your help

> You could modify [pretrain_datasets.json](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/configs/pretrain_datasets.json) and [...]

closed 2024-11-08 by prilgrim-whtie