thu-ml/RoboticsDiffusionTransformer #16: Mobile ALOHA Dataset Conversion Issues

WangZhiXiong0, 2024-11-07:
Thanks for sharing, I downloaded Mobile ALOHA Dataset from the official website, but I got the following error during the conversion process, it recognises the number of files but does not perform the conversion operation, but ends it directly. Can you help me?
[image]

WangZhiXiong0, 2024-11-08:
Thank you very much for your response and I will continue to look forward to follow-up instructions.

WangZhiXiong0, 2024-11-08:
I would like to ask you one more question. I would like to use only part of the RoboSet data to verify that I can run the code successfully. Apart from removing irrelevant datasets from pretrain_datasets.json and pretrain_sample_weights.json, are there any changes that need to be made to the ‘producers’ before I start pre-training?Thank you for your reply.

csuastt, 2024-11-09:
I am not sure whether TFDataset will raise an error if you only have part of the RoboSet files.

closed 2024-11-16 by ethan-iai

grainw, 2025-03-03:
The .hdf5 file I downloaded from the official website doesn't have the instruction key in it.Have you solved this problem? @WangZhiXiong0 @csuastt

LBG21, 2025-03-04:
We're currently re-uploading the Aloha dataset. We'll update the repo once it's done.