thu-ml/RoboticsDiffusionTransformer #19: t5-xxl is too large; can we replace it with the t5-base?

yizt, 2024-11-11:


ethan-iai, 2024-11-16:
Yes, it is possible to replace T5-XXL with T5-Base, but there are two important considerations to keep in mind:

1. The hidden dimensions may differ between T5-XXL and T5-Base, meaning that the adapter will need to be fine-tuned to ensure compatibility.
2. RDT-1B has been pre-trained with T5-XXL. Simply swapping it with T5-Base without additional alignment could result in a performance drop.

Please let me know if you’d like further clarification!

Wonder1905, 2024-12-09:
RDT-170m trained with different T5? (i.e not xxl)

ethan-iai, 2024-12-09:
To clarify, RDT-170m is indeed trained with T5-XXL. If you have any additional questions or need further details, please feel free to ask.

closed 2024-12-09 by ethan-iai