thu-ml/RoboticsDiffusionTransformer #20: Some doubts regarding fine-tuning

Henry-Ellis, 2024-11-11:
Thank you for your excellent contribution. I have fine-tuned the model of my mechanical arm with 170m and 1b, but the results are not satisfactory and the success rate is not high. I use single-arm data and two camera views, head and wrist, based on your suggestion, I only use right_arm, left_arm=0 and the image of the left wrist camera is the same as the image of the right wrist camera.I only have one task using a single instruction. I want to know where my problem lies and whether the fine-tuning process is reasonable. I look forward to your reply. 

The following images are my inference visualization
[image]

csuastt, 2024-11-12:
>  the image of the left wrist camera is the same as the image of the right wrist camera

Do not use the same image... Please PAD the left-wrist camera.

csuastt, 2024-11-12:
Yeah, your result is quite abnormal. I speculate that the training is not converged. Maybe you should train longer.

Besides, I have a few questions:
1. How many episodes are used?
2. What is the total batch size?
3. May I take a look at your loss curve?
4. Any other bugs?

Henry-Ellis, 2024-11-12:
> Yeah, your result is quite abnormal. I speculate that the training is not converged. Maybe you should train longer.
> 
> Besides, I have a few questions:
> 
> 1. How many episodes are used?
> 2. What is the total batch size?
> 3. May I take a look at your loss curve?
> 4. Any other bugs?

**Thank you for your reply, i will trt PAD later. Did you try the single-arm data when you fine-tuned it？** 
1.About 1000 episodes single task
2. --train_batch_size=4 \ --sample_batch_size=8 \
3.loss doesn't seem to be a problem、
4.Not yet
[image]

Very happy to follow your advice.

csuastt, 2024-11-12:
> Thank you for your reply, i will trt PAD later. Did you try the single-arm data when you fine-tuned it？

No, I didn't. But I don't think that is the point.

How many GPUs are used? I got that batch size per GPU is 4.

Henry-Ellis, 2024-11-12:
> How many GPUs are used? I got that batch size per GPU is 4.

Eight 4090 servers fine-tune the 1b model, A single 4090 server fine-tunes the 170m model

Henry-Ellis, 2024-11-14:
Thank you for your suggestions. The effect has improved a lot

closed 2024-11-14 by Henry-Ellis