thu-ml/RoboticsDiffusionTransformer #21: RuntimeError: mat1 and mat2 must have the same dtype, but got BFloat16 and Float

liushuailxx, 2024-11-12:
您好，非常感谢你们能将这么好的项目开源。
我在我们的机械臂上采集了示教数据，然后按照教程进行了微调，程序在运行到500步后，报错了。我是预先计算的语言编码，请问这个问题该如何解决。
[image]
期待您的回复

liushuailxx, 2024-11-12:
[image]

csuastt, 2024-11-12:
在sampling部分加个with autocast()吧，这个我们运行的时候不会遇到。我怀疑是和平台有关的问题。

liushuailxx, 2024-11-12:
> 在sampling部分加个with autocast()吧，这个我们运行的时候不会遇到。我怀疑是和平台有关的问题。

非常感谢，目前已经顺利运行了

closed 2024-11-12 by liushuailxx

Xubbbb, 2024-11-13:
这个错误是因为RDTRunner的lang_adaptor似乎没有设置成和lang_tokens一样的类型，在使用build_condition_adapter创建lang_adaptor时没有吧dtype设置好，所以lang_adaptor自动设置为了float32

csuastt, 2024-11-14:
> 这个错误是因为RDTRunner的lang_adaptor似乎没有设置成和lang_tokens一样的类型，在使用build_condition_adapter创建lang_adaptor时没有吧dtype设置好，所以lang_adaptor自动设置为了float32

有可能，你试过设置dtype吗？如果work可以提个pr😊