thu-ml/RoboticsDiffusionTransformer #22: Confusions Regarding Finetuning with New Dataset

JinmingM, 2024-11-12:
Thank you for sharing this outstanding work with the community. 
I have been working on fine-tuning the 1B model with a new dataset, but encountered some confusion:

1. In the configuration file, the "noise_scheduler" setting has **"prediction_type: sample"**, which corresponds to **"loss = F.mse_loss(pred, action_gt)" rather than "loss = F.mse_loss(pred, noise)"**. Since Diffusion Policy predicts the action rather than the noise, should this setting be modified in the configuration file?

2. During the fine-tuning process with a new dataset, should unused perspectives be padded with zeros? Also, what role does the "image_mask" in "dataset_img_keys.json" play in this process?

Look forward to your response. Thank you!

csuastt, 2024-11-12:
1. That is correct. In our setting, we predict the clean sample rather than the noise.
2. Pad it with a background image. You can simply return a zero-shape array like [here](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/data/hdf5_vla_dataset.py#L231). 
3. They are not used if you don't use TFdataset. But I am not sure whether there will be a bug if you don't fill in any values corresponding to your new dataset.

JinmingM, 2024-11-13:
Thanks for your answer! Nice work!

closed 2024-11-13 by JinmingM