thu-ml/RoboticsDiffusionTransformer #23: Confused about normalization

alik-git, 2024-11-15:
Hey folks, thanks for your great work.

I am confused about what normalization is happening. I understand in your README you say that you are not normalizing anything except the gripper state.

https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/README.md?plain=1#L122 

But at various places in the code there are state_std, state_mean, state_norm vectors being computed. To see them just do a search for "state_mean" in the repo. It's in [data/episode_transform.py](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/data/episode_transform.py#L280), [data/hdf5_vla_dataset.py](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/data/hdf5_vla_dataset.py#L98) etc. 

You also say to compute the dataset stats for making your own dataset.
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/docs/pretrain.md?plain=1#L230

What exactly are you using these for? Could you please explain and point me to the code where these are being used? It would be much appreciated, thank you.

ethan-iai, 2024-11-16:
1. `state_mean`: This is used as a dummy value during pretraining when states are randomly masked out. 
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/train/dataset.py#L277-L278
2. `state_std`: This serves as the base magnitude of noise applied to real states during training. Adding noise helps improve the model’s robustness.
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/train/dataset.py#L270-L273
3. `state_norm`: This is used during validation to calculate the value of the l2_loss as the denominator for normalization purposes. 
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/414715e34e6734ef4563f806114ff37752e0eb58/train/sample.py#L83
4. `state_min` & `state_max`: These are used for development purposes and are not essential for the core functionality.

alik-git, 2024-11-16:
super helpful, thank you so much!

closed 2024-11-16 by alik-git