thu-ml/RoboticsDiffusionTransformer #25: Pretrain Dataset Question:

motor-x, 2024-11-18:
Hi,

When you used BC Z (Jang et al., 2021) as one of your pretrain dataset, did you realize that their action data is problematic？ For example:

[image]

The gripper action for the future 10 steps is all 0, and the state of gripper increased; However, as shown below, the gripper action for the future 10 steps is also all 0, the state of gripper decreased.

[image]

Did you ever fixed this? Could you please share the fixed data? 

Thanks!

csuastt, 2024-11-18:
In pre-training, we do not use actions. We use future states instead.

motor-x, 2024-11-18:
Got it, thanks!

motor-x, 2024-11-19:
Also, could you please let me know where you got the ALOHA dataset? It seems that the dataset is missing from the google drive of open-x dataset: https://console.cloud.google.com/storage/browser/gresearch/robotics?pageState=(%22StorageObjectListTable%22:(%22f%22:%22%255B%255D%22))

Thanks!

motor-x, 2024-11-28:
Hi,

Just one further question, I noticed that you are also using RH20T dataset. Two questions about this:

- Where to find the state and action information? I noticed that they have:
  - join.npy
  - tcp_base.npy
  but, it seems that they are all state info, right?

- They have a number of camera views, may I ask which camera views did you use for pretrain?

Thanks!

LBG21, 2024-11-28:
We use both joint.npy and tcp_base.npy for future state prediction. To leverage the multiple camera views in RH20T, we randomly select one camera view from all views available for each data sample during training.

motor-x, 2024-11-28:
Thank you for the quick reply! Then, what about action? Are you using future joint and tcp_base as action?

Thanks!

LBG21, 2024-11-28:
During pre-training, we do not use action but instead future states. In the case of RH20T, the model takes current joint pose and tcp_base as proprio, and predicts future joint pose and tcp_base.

motor-x, 2024-11-28:
Got it, I appreciate your quick and detailed response, thank you!

motor-x, 2024-11-28:
Sorry for bothering you again! In RH20T, for the following two configs:

Cfg 1 Flexiv Dahuan AG95 OptoForce

LBG21, 2024-11-28:
During our pre-training processing, most (I'm not sure if it's all) of the joint.npy dimensions are 6 or 7, so we simply remove all data that is not 6 or 7 dimensions.

closed 2024-12-17 by ethan-iai