thu-ml/RoboticsDiffusionTransformer #28: I would like to inquire about robot vibrations while watching a video.

a510721, 2024-11-26:
Thank you for sharing your research results. 
In the video you uploaded, I noticed that the robot is vibrating.
What could be causing this phenomenon and can it be resolved?

csuastt, 2024-11-26:
This is caused by the hardware. This is resolved by switching to the ARX L6 robot arm.

a510721, 2024-11-27:
Can you please provide more details about what is wrong with the hardware (and its SDK)? 
Is there no way to solve the problem with the Aloha mobile robot?

ethan-iai, 2024-12-17:
> This is caused by the hardware (and its SDK). This is resolved by switching to the ARX L6 robot arm.

New ARX sdk, no vibrations 🥳

https://github.com/user-attachments/assets/340920f2-4910-4e61-8c5c-30d06b8d2068

closed 2024-12-17 by ethan-iai