thu-ml/RoboticsDiffusionTransformer #29: Fail to read dirty bit

wu-yutong-525, 2024-11-27:
Hi,  I try to finetune rdt-170m using Berkeley-UR5 dataset from the pretraining mixture following this [README](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/docs/pretrain.md). After I begin finetuning by running the `pretrain.sh`with revised parameters following the instructions, I continuously encounter the `Fail to read dirty bit` issue. Is it normal?

JayceNing, 2024-11-28:
I met the same problem, did you solve the problem?

[image]

csuastt, 2024-11-29:
As in Readme, you should first run the producer. AFTER the filling is completed, run your training script...

curryqka, 2024-12-11:
> As in Readme, you should first run the producer. AFTER the filling is completed, run your training script...

I'd like to know how to tell if the producer script is complete. Looking forward to your kindly reply.

csuastt, 2024-12-11:
@curryqka : When the worker prints the dirty ratio, the filling-up is completed.

closed 2024-12-17 by ethan-iai