thu-ml/RoboticsDiffusionTransformer #30: Questions about open-source fine-tuning datasets

JayceNing, 2024-11-28:
First of all, I would like to express my sincere appreciation for your open-source work and the contribution it makes to the community.

I have downloaded the open-source dataset and followed the command provided to extract the data. `cat rdt_data.tar.gz.* | tar -xzvf -
`
However, I encountered an error stating that the dataset is incomplete. I checked the number and size of the compressed files, and visually there doesn't seem to be any issue. The download also completed without errors. Could you kindly advise on how to resolve this issue?
[image]

Thank you for your time and assistance.

Best regards

csuastt, 2024-11-29:
Thank you for your attention. It seems that the download is incomplete due to network problems.

JayceNing, 2024-11-29:
Thank you for your response. I appreciate your help. It seems that the download was indeed incomplete due to network issues. I will try downloading the dataset again and ensure the network connection is stable.

closed 2024-11-29 by JayceNing

liangyuxin42, 2024-12-18:
@JayceNing I am experiencing the same problem. May I ask have you solved the problem after re-downloading?

JayceNing, 2024-12-18:
After re-downloading and extracting, it still throws an error at the end, but most of the data has already been extracted. When using ```data/hdf5_vla_dataset.py``` to read the dataset, an error occurs when reading the ```rdt_data/write_board_13/episode_1.hdf5``` file, which should be deleted. Additionally, the ```expanded_instruction_gpt-4-turbo.json``` file should be added to the ```write_board_13``` folder, following the format of other ```write_board_``` tasks. After this processing, there are 6109 usable episodes.

> @JayceNing I am experiencing the same problem. May I ask have you solved the problem after re-downloading?

liangyuxin42, 2024-12-18:
> ile should be added to the `write_board_13` folder, following the format of other `write_board_` tasks. After this processing, there are 6109 usable episodes.

Thanks for the reply, saved me a lot of time! :)

Luo-Zhongwei, 2025-07-20:
cool.  I  meet the same problem  , thanks for your  exp