thu-ml/RoboticsDiffusionTransformer #35: missing 'qpos' in bridgeV2 dataset

huangxu1991, 2024-12-02:
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/f6a57c2eaaa7b4fdb2ac7432f30ea926029b86df/data/preprocess_scripts/bridgev2.py#L131-L132

hi, it seems that KEY 'qpos' is missing.

LBG21, 2024-12-11:
Hi, if you have followed our instructions and executed `data/bridgev2/bridgedata_raw_to_numpy.py` and `data/bridgev2/bridgedata_numpy_to_tfrecord.py` to convert the raw data into TFRecord format, the resulting `.tfrecord` files should contain `qpos`. Could you please share the error message or relevant code snippets?

huangxu1991, 2024-12-17:
thx a lot. I'll check the procedure first.

ethan-iai, 2024-12-17:
> thx a lot. I'll check the procedure first.

Please feel free to let me know if you have any updates. I’d be happy to assist further if needed! 😊

closed 2024-12-18 by ethan-iai