thu-ml/RoboticsDiffusionTransformer #37: Some question about Finetuning with New Dataset

shiqitannn, 2024-12-05:
Thank you for sharing the  great research. I have been working on fine-tuning the 1B model with a new dataset,  my setting is:
- number of train episodes: 367
- number of validation episodes: 41
- a single task with a fixed instruction
- batch size 8,  train on 8 A800,  so the total batch size is 64
- train steps: 10w 
I found that the model fits very well on the training set, but performs poorly on the validation set and poorly when deploy in real robots.
[image]
[image]

 I want to know where my problem lies and whether the fine-tuning process is reasonable. I look forward to your reply.

shiqitannn, 2024-12-05:
And I would like to ask a question regarding this issue: On the A800, when I increased the batch size from 8 to 32, the training time also increased fourfold. Is this normal?

lanlankilkil, 2024-12-10:
I'd like to ask what type of robotic arm you trained with, and is it necessary to collect so many episodes for a single task，and ow long have you been training at 8-a800?

csuastt, 2024-12-10:
Have you considered using a much smaller model, like RDT-170m?

Btw, we will release our simulation benchmark within a week. You may have a try as a reference.

ethan-iai, 2024-12-17:
After private discussion, this issue has been resolved.

closed 2024-12-17 by ethan-iai

lanlankilkil, 2025-01-02:
Hello, I would like to inquire about some questions regarding RDT. Could we exchange contact information?

csuastt, 2025-01-02:
@lanlankilkil https://rdt-robotics.github.io/rdt-robotics/wechat.html

chenyt31, 2025-01-31:
The WeChat group QR code has expired. Would you mind sending the group again? @csuastt

csuastt, 2025-02-01:
@chenyt31 

[image]