thu-ml/RoboticsDiffusionTransformer #38: Positional Encoding Questions

motor-x, 2024-12-05:
Hi

I was wondering why you prefer to learnable positional encoding for state, rather than using sinusoid encoding (used in like ACT).

self.x_pos_embed = nn.Parameter(torch.zeros(1, horizon+3, hidden_size))

Thanks

csuastt, 2024-12-10:
All the positional embeddings are learnable. What you mentioned is just an initialisation.

motor-x, 2024-12-12:
Thank you for the reply! I know you are using learnable positional embeddings. **My question was: why not using sinusoid encoding?** Did you see any absolute advantages of using learnable ones?

Thanks!

csuastt, 2024-12-12:
> Did you see any absolute advantages of using learnable ones?

Sry:( We haven't conducted any ablation studies on this. But I believe sinusoid encodings would be good enough...