thu-ml/RoboticsDiffusionTransformer #42: Wonderful Work!! A few questions regarding modelling and simulation tasks

gaodechen, 2024-12-11:
Hi,

Thank you for your fantastic work and for making the codes and checkpoints publicly available! I was quite surprised by the trianing stability and its generalizaibility to simulation environments though it seems the pre-trianing does not include sim data.

I’m currently fine-tuning RDT-1B and testing it on several simulation tasks in [AV-ALOHA](https://soltanilara.github.io/av-aloha/). I would greatly appreciate your insights on a few questions:

* **Action Jerkiness:** RDT-1B seems to use 64 as the default chunk size. Have you encountered any jerkiness between action chunks, as shown in the attached video (e.g., the robotic arm suddenly moving to an unintended position at 4s)? 
* **Fine-Tuning Steps:** Based on your extensive experiments, how many demonstrations or training steps are typically required to achieve a reasonable policy for few-shot learning of downstream tasks? I understand RDT-1B is able to learn solely with 5 or 6 demos as mentioned by the paper. I'm currently reusing the finetune.sh scripts and it took around 200 steps (bs=16, #GPU=4) for the loss to converge for 50 demos while the policy was kinda jerky. So just wondering if there were separate hyper-parameters for your few-shot experiments based your emprical results. 
* **Model Choice:** Could you share the rationale for selecting T5 as the backbone instead of other LLMs?

Looking forward to your inputs. Thanks again for this amazing contribution!

[image]

https://github.com/user-attachments/assets/6ff95be6-103d-45ba-91f5-cb45a5fc01af

gaodechen, 2024-12-11:
p.s. just got finetuning work for a downstreaming task with 50 demos and the hyper-parameters were batch size = 32 and training steps >= 400 (didn't expect this since they were quite large for the dataset scope? idk..) when # GPU = 4

https://github.com/user-attachments/assets/edc8d9c3-793c-4aa6-b72b-56f2924dcaef

LBG21, 2024-12-28:
Hi, we’ve tested RDT-1B on the ManiSkill Simulation Benchmark, and the results have been released at [image]. We’d like to share some insights on testing RDT in simulation environments based on your questions: 

1. **Action Jerkiness**: RDT has not exhibited any noticeable jerkiness in either simulation or real-world experiments. The choice of chunksize can depend on the length of your training episodes. For shorter episodes (e.g., 100–200 frames), you might try reducing the chunksize through action interpolation. For instance, in ManiSkill, we use 4x action interpolation, resulting in an effective chunksize of 16, as the average training episode length is approximately 100 frames.

2. **Fine-Tuning Steps**: In simulation, we recommend using a higher number of training episodes to address the sim-to-real gap between pretraining and fine-tuning. For example, we use 1,000 trajectories per task in ManiSkill to fine-tune our model. Regarding training steps, RDT was trained on ManiSkill with 5,000 trajectories for 150k steps. A few hundred training steps are usually insufficient for RDT to perform effectively, as more training steps often lead to better performance.

3. **Language Encoder**: We use T5 because it is a widely adopted and effective choice for language encoding.

gaodechen, 2024-12-30:
> Hi, we’ve tested RDT-1B on the ManiSkill Simulation Benchmark, and the results have been released at [image]. We’d like to share some insights on testing RDT in simulation environments based on your [...]

Thanks again for the wonderful work and for sharing the insights! Closed the issue.

closed 2024-12-30 by gaodechen