thu-ml/RoboticsDiffusionTransformer #52: Is it possible to run RDT directly in a simpler environment without additional fine-tuning?

Koorye, 2024-12-21:
Thank you for your excellent work! I tried to migrate the RDT to simplerenv, but it didn't work as expected and the robot moved in the wrong direction.
Google robot and widowx are available in simplerenv, along with an EEF delta-based controller, so I tried the following: 
1. Set the RDT proprio input to pos (x, y, z) and 6d-rot.
2. Get the eef pose from SimplerEnv, then convert into pos (x, y, z) and 6d-rot as proprio input to the RDT.
3. Convert the RDT output action sequence into pos (x, y, z) and euler angles (pitch、yaw、roll), and subtract from the previous action to get eef delta.
Thank you for the virtual environment, however, it does not fully meet my requirement, I wanted to test the pre-training model directly in a virtual environment without fine-tuning, is it possible to achieve?

ethan-iai, 2024-12-22:
Thank you for your question! Predicting delta values may indeed go beyond the original scope of RDT pre-training, as it was primarily trained to predict the absolute target end-effector (EEF) or joint positions. On the other hand, velocity predictions, which are more closely related to delta predictions, were not as extensively trained.

Given this, I would recommend further adapting RDT to the delta action space through additional fine-tuning. This would help improve its performance in predicting delta values. For more details, please refer to #14.

I hope this helps, and feel free to reach out if you have more questions!

Koorye, 2024-12-23:
Thanks for your answer! I may not have made myself clear:
1. I use the api of SimplerEnv to get the absolute eef position `(right_eef_pos_x,right_eef_pos_y,right_eef_pos_z,right_eef_angle_0,right_eef_angle_1,right_eef_angle_2,right_eef_angle_3,right_eef_angle_4,right_eef_angle_5)` as model input.
2. After the model predicts the sequence of eef absolute position, I subtract each eef from the previous eff to get the eef delta `(eef_delta=curr_eef - prev_eef)`.
Maybe I will try further fine-tuning to get better adaptation, but I also want to know whether there are problems with the above approach?

ethan-iai, 2024-12-23:
That sounds promising! However, I’m not very familiar with SimplerEnv, so there might be some bugs that I’m not fully aware of.

Koorye, 2024-12-23:
Thank you! I'll contact you if I have any questions!

closed 2024-12-23 by Koorye