thu-ml/RoboticsDiffusionTransformer #53: How to find the path of CUTLASS?

Intellouis, 2024-12-22:
Hi! Thanks for your great work! I have a question about ```cutlass PATH```.

As shown in the README, we should modify some of the parameters in```finetune.sh```, such as ```CUTLASS_PATH``` and ```WANDB_PROJECT```. I wonder how I could find the proper path of ```cutlass``` on my machine.

When I run ```find -n "*cutlass*"``` under ```/```, I can only get:
```
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/inference/v2/kernels/cutlass_ops
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/inference/v2/kernels/cutlass_ops/cutlass_ops.cpp
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/inference/v2/modules/implementations/moe/cutlass_multi_gemm.py
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/inference/v2/modules/implementations/moe/__pycache__/cutlass_multi_gemm.cpython-310.pyc
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/ops/op_builder/inference_cutlass_builder.py
/root/anaconda3/envs/rdt/lib/python3.10/site-packages/deepspeed/ops/op_builder/__pycache__/inference_cutlass_builder.cpython-310.pyc
/root/anaconda3/pkgs/pytorch-2.5.1-py3.10_cuda12.4_cudnn9.1.0_0/lib/python3.10/site-packages/torch/_inductor/codegen/cuda/__pycache__/cutlass_epilogue_gen.cpython-310.pyc
/root/anaconda3/pkgs/pytorch-2.5.1-py3.10_cuda12.4_cudnn9.1.0_0/lib/python3.10/site-packages/torch/_inductor/codegen/cuda/__pycache__/cutlass_utils.cpython-310.pyc
/root/anaconda3/pkgs/pytorch-2.5.1-py3.10_cuda12.4_cudnn9.1.0_0/lib/python3.10/site-packages/torch/_inductor/codegen/cuda/cutlass_epilogue_gen.py
/root/anaconda3/pkgs/pytorch-2.5.1-py3.10_cuda12.4_cudnn9.1.0_0/lib/python3.10/site-packages/torch/_inductor/codegen/cuda/cutlass_lib_extensions
/root/anaconda3/pkgs/pytorch-2.5.1-py3.10_cuda12.4_cudnn9.1.0_0/lib/python3.10/site-packages/torch/_inductor/codegen/cuda/cutlass_utils.py
```

I tried to search "how to find my path to cutlass" on google, still with no corresponding answers.

Should I manually install something related to cutlass, which is not mentioned in the setup phase in README? If so, what should I do exactly? Thanks for you help!

Heron0625, 2024-12-22:
i encountered the same issue and solved by installing it referring to https://blog.csdn.net/gitblog_00440/article/details/142837443?utm_medium=distribute.pc_relevant.none-task-blog-2~default~baidujs_baidulandingword~default-1-142837443-blog-123019605.235^v43^pc_blog_bottom_relevance_base2&spm=1001.2101.3001.4242.2&utm_relevant_index=4

Intellouis, 2024-12-22:
> i encountered the same issue and solved by installing it referring to [...]

It seems that I should clone the repo ```github.com/NVIDIA/cutlass``` and make my ```CUTLASS_PATH``` linked to this cloned directory. Thanks!

closed 2024-12-22 by ethan-iai