thu-ml/RoboticsDiffusionTransformer #58: rdt-170M fine-tuning

yi-shui-han, 2024-12-25:
Hello, I have a question. When I was testing the fine-tuning of maniskill data, I selected text_embed_PegInsertionSide-v1.pt as the language embeddings, copied the hdf5_maniskill_dataset.py code to hdf5_vla_dataset, and executed the finetune_maniskill.sh script. However, after importing the file during training, the dimension of lang_embed was three, and after padding with pad_sequence, it became four-dimensional. Therefore, when indexing L, the dimension does not match. What is the reason?
[image]
[image]

ethan-iai, 2024-12-27:
@LBG21

LBG21, 2024-12-28:
Hi, the .pt files are used for inference and are not tested during fine-tuning (even though their values are the same). I believe they include an extra batch dimension. You can directly remove the 0th dimension (which should be 1) from `lang_embed` during fine-tuning.

closed 2025-01-03 by yi-shui-han