thu-ml/RoboticsDiffusionTransformer #59: RuntimeError: mat1 and mat2 shapes cannot be multiplied (200x768 and 4096x1024)

yi-shui-han, 2024-12-25:
thank you very much for your open source project.
When I fine-tuned the `maniskill data`.  I replaced `t5-v1_1-xxl` with `t5-v1_1-base`, and then an error occurred during fine-tuning. How can I solve this problem? Looking forward to your reply.

`RuntimeError: mat1 and mat2 shapes cannot be multiplied (200x768 and 4096x1024)`

ethan-iai, 2024-12-27:
You should update the `lang_token_dim` in config.
https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/947a0f6613f8a1aa13cf63b0a772ca66a9f4228f/configs/base.yaml#L44

closed 2024-12-27 by ethan-iai

LEE23Xuanyi, 2025-02-13:
hello, may i ask, how did you replace xxl with base. Except for encode_lang_batch.py, what else did you changed.

this is what my .py output:

(rdt) george@george-ERAZER-H610M-WIFI:/data/23XuanYi/RoboticsDiffusionTransformer$ /home/george/.conda/envs/rdt/bin/python /data/23XuanYi/RoboticsDiffusionTransformer/scripts/encode_lang_batch.py
You are using the default legacy behaviour of the <class 'transformers.models.t5.tokenization_t5.T5Tokenizer'>. This is expected, and simply means that the `legacy` (previous) behavior will be used so nothing changes for you. If you want to use the new behaviour, set `legacy=False`. This should only be set if you understand what it means, and thoroughly read the reason why this was added as explained in https://github.com/huggingface/transformers/pull/24565
/home/george/.conda/envs/rdt/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated. It will be removed in the future and UntypedStorage will be the only storage class. This should only matter to you if you are using storages directly.  To access UntypedStorage directly, use tensor.untyped_storage() instead of tensor.storage()
  return self.fget.__get__(instance, owner)()
WARNING:root:Some parameters are on the meta device device because they were offloaded to the disk.
0it [00:00, ?it/s]

i've made sure that my path had nothing wrong
the path in ,py is set like:

MODEL_PATH = "google/t5-v1_1-large"
CONFIG_PATH = "configs/base.yaml"
# Modify the TARGET_DIR to your dataset path
TARGET_DIR = "data/datasets/task_n100/tfrecords"

# Note: if your GPU VRAM is less than 24GB, 
# it is recommended to enable offloading by specifying an offload directory.
OFFLOAD_DIR = "/data/23XuanYi/offload" # Specify your offload directory here, ensuring the directory exists.

and this is the path of my .json 
/data/23XuanYi/RoboticsDiffusionTransformer/data/datasets/task_n100/tfrecords/placecup/expanded_instruction_gpt-4-turbo.json

yi-shui-han, 2025-02-14:
I'm fine-tuning the model, so the size of the model has been fixed, and I can't replace the xxl with base, if you're doing pre-training, you can try to replace it, but if you just fine-tune, you can choose to text_embed it first, and then fine-tune it