thu-ml/RoboticsDiffusionTransformer #61: Deployment on Different Robots

wdi-nancy, 2024-12-26:
Has the author successfully tested tasks on robotic arms other than ALOHA, given that RDT supports various robots?

csuastt, 2024-12-28:
In the simulator, we have deployed our model on the Franka robot. For real-world robots, the community members have successful deployments on UR, Kinova, and so on. There will probably be a PR for Kinova soon.

wdi-nancy, 2025-01-02:
> In the simulator, we have deployed our model on the Franka robot. For real-world robots, the community members have successful deployments on UR, Kinova, and so on. There will probably be a PR for Kinova soon.

Great! These two robots should also be controlled by joints. Have there been any successful deployments on robots controlled by EEF? Considering this serves as a validation of the Unified Action Space.

wdi-nancy, 2025-01-02:
> In the simulator, we have deployed our model on the Franka robot. For real-world robots, the community members have successful deployments on UR, Kinova, and so on. There will probably be a PR for Kinova soon.

Great! These two robots should also be controlled by joints. Have there been any successful deployments on robots controlled by EEF? Considering this serves as a validation of the Unified Action Space. And have you tested the deployment performance in a single-arm scenario, using only the right-arm camera?

csuastt, 2025-01-02:
The community have successful deployments on the single-arm robot. The single-arm is much easier than the dual-arm. EEF is also OK. We have tried EEF control in our ongoing work, but it hasn't been released yet.

yzzueong, 2025-01-14:
> The community have successful deployments on the single-arm robot. The single-arm is much easier than the dual-arm. EEF is also OK. We have tried EEF control in our ongoing work, but it hasn't been released yet.

Hello, could you provide a link about the single-arm robot deployment? Thanks