thu-ml/RoboticsDiffusionTransformer #62: Failure about unzipping the demo_1k.zip

ergo-zyh, 2024-12-28:
Dear, I fail when I unzip the demo_1k.zip after I run the command  _cat demo_1k_part_* > demo_1k.zip_. And the error appears like
Archive:  demo_1k.zip
fatal error: read failure while seeking for End-of-centdir-64 signature.
  This zipfile is corrupt.
unzip:  cannot find zipfile directory in one of demo_1k.zip or
        demo_1k.zip.zip, and cannot find demo_1k.zip.ZIP, period.

Do you meet the problem? Is there any wrong thing or I run it wrong?
Thanks

LBG21, 2024-12-28:
Hi, we're currently reviewing the files to ensure if there's any file missed. We should have this resolved shortly and we'll update as soon as possible.

ergo-zyh, 2024-12-28:
> Hi, we're currently reviewing the files to ensure if there's any file missed. We should have this resolved shortly and we'll update as soon as possible.

Thanks a lot.

LBG21, 2024-12-29:
Hi, it appears that the files are correct and complete. You can try re-downloading these files.

ergo-zyh, 2025-01-03:
Thank you for your help. I have verified the SHA256 of each file, and they are all correct. I'm unsure about what could be causing the issue.

LBG21, 2025-01-05:
I'll try re-upload these files.

Jimntu, 2025-06-19:
Hi, a follow-up question. Have you managed to unzip the files? If so, are they in rlds format? Thanks!