thu-ml/RoboticsDiffusionTransformer #67: How to record my custom finetune data?

seeingrain, 2025-01-13:
So, the code or paper does not mention how to record custom finetune dataset.
Does anyone know what tool should I use to record my own data?

seeingrain, 2025-01-13:
Should I refer to [https://github.com/MarkFzp/mobile-aloha/blob/0e403249c76054a68e757e590d4da4dba401c9e3/aloha_scripts/record_episodes.py](url) from MobileALOHA project?

csuastt, 2025-04-05:
The recording script depends on your platform. In our case (AGILEX bi-manual robot), we refer to https://github.com/agilexrobotics/cobot_magic/blob/main/collect_data/collect_data.py. And our pipeline totally supports HDF5 format. See  https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/README.md#Fine-Tuning-on-Your-Own-Dataset and https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/data/hdf5_vla_dataset.py.

closed 2025-04-05 by csuastt