thu-ml/RoboticsDiffusionTransformer #73: missing language instructions to covert mobile aloha dataset

fayefw, 2025-01-20:
Hi, thank you for your great work! I tried to covert the mobile aloha dataset from hdf5 to tfrecords but the output was empty. I inspected the  hdf5 file and found there is no instruction in it. could you tell me how can I find or set the dataset with instructions? thanks a lot!

thkkk, 2025-01-24:
You can check the `expanded_instruction_gpt-4-turbo.json` in the dataset, which is the instructions of the tasks. During training, one of the three prompts will be uniformly randomly sampled. If the long prompt is selected, the algorithm will also uniformly randomly select a prompt from the long prompt.

xiaolanliu, 2025-04-22:
> You can check the `expanded_instruction_gpt-4-turbo.json` in the dataset, which is the instructions of the tasks. During training, one of the three prompts will be uniformly randomly sampled. If the long prompt is selected, the algorithm will also uniformly randomly select a prompt from the long pr
Thanks for your reply,may i ask where to generate this json file?

103240, 2025-04-23:
> 嗨，感谢您的出色工作！我尝试将移动 aloha 数据集从 hdf5 转换为 tfrecords，但输出为空。我检查了 hdf5 文件，发现其中没有任何说明。您能告诉我如何根据说明查找或设置数据集吗？多谢！
请问您有解决办法了吗，我也想做这个转换