thu-ml/RoboticsDiffusionTransformer #84: Mobile ALOHA Dataset Conversion Issues

grainw, 2025-03-03:
The .hdf5 file I downloaded from the official website doesn't have the instruction key in it.

LBG21, 2025-03-04:
We're currently re-uploading our processed aloha dataset. We'll update this repo once it's done.

wojiaohumaocheng, 2025-03-05:
meet same problem

Bobo-y, 2025-03-12:
> We're currently re-uploading our processed aloha dataset. We'll update this repo once it's done.

hello, can you just upload all json files in one zip or tar file? may be we don't want to download all data again.

LBG21, 2025-03-13:
> > We're currently re-uploading our processed aloha dataset. We'll update this repo once it's done.
> 
> hello, can you just upload all json files in one zip or tar file? may be we don't want to download all data again.

The size of our processed Mobile Aloha dataset is approximately ~1TB, therefore we cannot provide all files in one zipped file due to huggingface constraints.

closed 2025-04-05 by csuastt