thu-ml/RoboticsDiffusionTransformer #85: why 0% success when I run eval_rdt_mani

Un1Lee, 2025-03-10:
I run : python -m eval_sim.eval_rdt_maniskill --pretrained_path ~/rl_ws/RoboticsDiffusionTransformer/maniskill-model/rdt/mp_rank_00_model_states.pt 

Then there are some warning, and Loading weights from pretrained_path.
The bar goes, and finally Success rate: 0.0%.

Why?

LBG21, 2025-03-13:
Can you provide more details? e.g. You could record a inference video in simulation?

Royhuiy, 2025-03-25:
> I run : python -m eval_sim.eval_rdt_maniskill --pretrained_path ~/rl_ws/RoboticsDiffusionTransformer/maniskill-model/rdt/mp_rank_00_model_states.pt
> 
> Then there are some warning, and Loading weights from pretrained_path. The bar goes, and finally Success rate: 0.0%.
> 
> Why?

i get the same result, did you 
Have you solved this problem?

Royhuiy, 2025-03-25:
> Can you provide more details? e.g. You could record a inference video in simulation?

hi, i get the same resutl , But I don't know how to record a inference video in simulation? Can you provide the corresponding code?

csuastt, 2025-04-04:
You can try to git pull the repo. We have fixed some bugs before.

roychen1998, 2025-10-28:
I have a similar issue. When global_steps greater or equal to 50, truncated=1, but info['success']=false.

https://github.com/user-attachments/assets/b878971e-8398-4e25-bbee-6559927aebe3

It seems that the policy acts without regarding the cube position, and it always stops when global_steps == 50.