thu-ml/RoboticsDiffusionTransformer #89: Question about pretraining dataset

J-Oyasumi, 2025-03-20:
Thanks for your excellent work! I'm wondering if the mobile aloha dataset you provide on huggingface contains the aloha dataset? If not, where can I get the intact aloha dataset since there is not 'instruction' key in the dataset downloaded from the official website.

closed 2025-03-21 by J-Oyasumi

Luo-Zhongwei, 2025-07-20:
> Thanks for your excellent work! I'm wondering if the mobile aloha dataset you provide on huggingface contains the aloha dataset? If not, where can I get the intact aloha dataset since there is not 'instruction' key in the dataset downloaded from the official website.

hi, I am a new learner , how do you solve your problem

J-Oyasumi, 2025-07-20:
> hi, I am a new learner , how do you solve your problem

the released mobile aloha dataset contains the aloha dataset. There is a `aloha_static` folder(I don't remember the exact name) which stores the aloha dataset. :)

Luo-Zhongwei, 2025-07-20:
> > > 感谢您的出色工作！我想知道您在 huggingface 上提供的移动 Aloha 数据集是否包含 Aloha 数据集？如果没有，我该在哪里获取完整的 Aloha 数据集？因为从官方网站下载的数据中没有“instruction”键。
> > 
> > 
> > 你好，我是新学员，请问你的问题是怎么解决的
> 
> 已发布的移动版 Aloha 数据集包含 Aloha 数据集。有一个`aloha_static`文件夹（我不记得具体名称了）存储着 Aloha 数据集。:)

does the rdf-ft data in huggingface equal to real aloha dataset?