thu-ml/RoboticsDiffusionTransformer #92: Changing the instructions does not change policy

destin-v, 2025-03-21:
I modified `eval_sim.eval_rdt_maniskill.py` by inserting my own custom instruction to see if the policy would accept different instructions.  What I found out was that the policy is unaffected by changes to the instructions.

```python
task2lang = {
    "PegInsertionSide-v1": "Pick up a orange-white peg and insert the orange end into the box with a hole in it.",
    # "PickCube-v1": "Grasp a red cube and move it to a target goal position.",
    "PickCube-v1": "Do nothing.",   # CUSTOM INSTRUCTION
    "PickSingleYCB-v1": "Pick up the item and move it to the target goal position.",
    "StackCube-v1":  "Pick up a red cube and stack it on top of a green cube and let go of the cube without it falling.",
    "PlugCharger-v1": "Pick up one of the misplaced shapes on the board/kit and insert it into the correct empty slot.",
    "PushCube-v1": "Push and move a cube to a goal region in front of it."
}
```

This seems odd, since I would expect that modifications to the instructions would change its policy.  This seems like a bug.

csuastt, 2025-04-04:
This is normal. Because the language diversity of the trajectory is insufficient, the model cannot learn rich semantics. If there are different tasks guided by different languages ​​in the same scene, then the language can play a role. Consider a table scene, "pick up apples" corresponds to the task of picking up apples, and "pick up bananas" corresponds to the task of picking up bananas. At this time, the model can perform specific tasks based on a specific language. The simulator example we gave is too simple, and it is enough to distinguish different tasks based on images alone. Thus, it is reasonable that the model cannot learn good language representations.

destin-v, 2025-04-04:
This appears to be somewhat misleading as the project is showcasing using language to control the robot.  If language plays no role in the task and the robot is just using the image to perform the task what is the point of adding in language at all?

csuastt, 2025-04-05:
This only happened in our ManiSkill simulation benchmark, as it is NON-LANGUAGE benchmark. You can find the language-following capability in our real-world experiments. If you want to test language capability, you can try language-conditioned benchmarks like LIBERO rather than non-language benchmarks like ManiSkill.

closed 2025-04-05 by csuastt

destin-v, 2025-04-05:
Maniskill provides an environment called PickSingleYCB-v1 which provides a randomized set of objects in front of the robotic arm.  So it is possible to use language to pick out different objects.  So you can use language like "pick up the apple" or "pick up the banana".  However, even when I use the PickSingleYCB-v1 environment and change the language commands, it does not work.  This leads me to think that these examples are quite brittle and difficult for general use.

csuastt, 2025-04-06:
You need to fine-tune on PickSingleYCB-v1 first. It seems the ManiSkill does not provide demonstrations. You need to generate the imitation learning dataset first and then fine-tune the model. See https://huggingface.co/datasets/haosulab/ManiSkill_Demonstrations/tree/main