thu-ml/RoboticsDiffusionTransformer #95: 致开发者：急需单个情节数据的完整示例 | Request to Developers: Urgently Need a Comprehensive Example of a Single Episode's Data

b4r4d41z, 2025-04-02:
尊敬的开发者，

我们迫切需要一个完整且直观的单个episode数据示例。

是否有小于500GB的示例数据集可供使用？ 正在构建自己数据集的研究人员特别希望了解episode的结构及其中的数据格式。 仅凭readme.md的描述和[Hugging Face](https://huggingface.co/datasets/robotics-diffusion-transformer/rdt-ft-data)上提供的信息，很难理解数据结构。特别是 `instruction`、`expanded_instruction` 和 `simplified_instruction` 字段的内容尚不明确。​

非常需要您的帮助！
______________________________________________________________________________________________________________________________________________________________
Dear Developers,

We urgently require a comprehensive and illustrative example of a single episode's data.

Are there sample datasets available that are significantly smaller than 500GB? Researchers compiling their own datasets are particularly interested in understanding the structure of episodes and the data formats they encompass. It's challenging to grasp the data structure solely based on the readme.md description and the information provided on [Hugging Face](https://huggingface.co/datasets/robotics-diffusion-transformer/rdt-ft-data) . Specifically, the contents of `instruction`, `expanded_instruction`, and `simplified_instruction` fields are unclear.​

Your assistance is greatly needed!

LBG21, 2025-04-07:
Sure. We'll be uploading a single episode from our finetune dataset soon.

b4r4d41z, 2025-04-07:
By the way, I’ve been wondering about this for a long time, and today I finally found the answers. I’ve written a detailed description of the dataset structure in this [issue](https://github.com/thu-ml/RoboticsDiffusionTransformer/issues/81)

In any case, thanks to the  for not leaving this issue without attention. I’m sure that having a small but complete data sample will be truly helpful for others in the future

closed 2025-04-07 by b4r4d41z