thu-ml/RoboticsDiffusionTransformer #102: finetuning on libero

lzl2040, 2025-04-24:
When I try to finetune RDT on LIBERO, the loss curve is as follows:

[image]
Is this normal? My dataset preparation is as follows:

- complete the ```libero_vla_dataset.py``` based on [hdf5_maniskill_dataset.py](https://github.com/thu-ml/RoboticsDiffusionTransformer/blob/main/data/hdf5_maniskill_dataset.py): Because the libero provides the actions and the  states in the format of ```x, y, z, roll, pitch, yaw, gripper```, I first transform it to [6D representation](https://arxiv.org/pdf/1812.07035). Then I use the following state:
```python
UNI_STATE_INDICES = [ STATE_VEC_IDX_MAPPING['eef_pos_x']
                            ] + [ STATE_VEC_IDX_MAPPING['eef_pos_y']
                            ] + [ STATE_VEC_IDX_MAPPING['eef_pos_z']
                    ] + [
                        STATE_VEC_IDX_MAPPING[f"eef_angle_{i}"] for i in range(6)
                    ] + [
                        STATE_VEC_IDX_MAPPING["right_gripper_open"]
                    ]
```
Is there any error?

jasper0314, 2025-08-19:
Hi @lzl2040 and @Buming-O , could you please share the performance of RDT on LIBERO that you’ve tried? I’m considering using RDT, but I’m not sure how well it works in simulation environments like LIBERO or SimplerEnv. Thanks!

Zctoylm0927, 2026-01-20:
> Hello! Thanks for your work. I'm also trying to finetuning LIBERO (using the original HDF5 version instead of the RLDS version provided by OpenVLA), and here are the current results. Can this be considered convergence?
> 
> [image]

Great try, could you please share the performance of RDT on LIBERO? I also encountered some problems during training.