thu-ml/RoboticsDiffusionTransformer #103: Question about the fine-tuned checkpoint.

zql-kk, 2025-04-28:
Thank you for a great work! Do you plan to release the fine-tuned checkpoint?

csuastt, 2025-06-04:
Thanks for your interest! 🙏 We are not planning to release our original checkpoint due to intellectual property concerns. Feel free to train your own using the base model and code in the repo!​

Luo-Zhongwei, 2025-07-20:
> Thanks for your interest! 🙏 We are not planning to release our original checkpoint due to intellectual property concerns. Feel free to train your own using the base model and code in the repo!​

hey  I actually don't understand what is the meaning of fine-tuned ckpt   , author already  release  "robotics-diffusion-transformer/maniskill-model"  and  "robotics-diffusion-transformer/rdt-1b"   ,  Aren’t these two fine‑tuned checkpoints?

csuastt, 2025-07-21:
@Luo-Zhongwei The one fine-tuned on the bimanual dataset...